fossasia/badgemagic-hardware #19: Make the battery detachable without the need of soldering - have to be detached to update firmware at the moment

froscon2024, 2024-09-22:
At the moment the battery have to be detached to update firmware.
https://github.com/fossasia/badgemagic-firmware/issues/38
Having to apply high temperature at the contacts of a lithium battery to install FOSS-firmware is really not great. Please make the battery detachable with a typical connector. This would also make it more user friendly to replace or upgrade the battery. Soldered in batteries are in general not great and if the badge would at some time be reviewed by ifixit it would get negative points for the soldered in battery. There are a ton of cheap, simple and flat connectors you can use instead.

kienvo, 2024-09-23:
Not sure how ifixit will rate a 2-pos DIP switch for both this and #18 :D

froscon2024, 2024-09-25:
I was more thinking of a connector for the battery. Not a DIP switch. Ifixit love of course not soldered in batteries. You can disconnect the battery thanks to the connector to flash but also to hack around with the badge. When you for example need for a special use case where the badge should run a week untouched, then disconnect the internal battery, leave the back off and connect with the same connector type some different battery. For example a 5000mAh 21700 battery cell.
Typical connectors hardware hackers and hackerspaces have around are the ones from the CR2032 batteries used in thinkpads. The connector is small, should fit inside the badge without problems and hardware hackers can already have some of them around.
[image]

PR #24 (2024-11-03): feat: detachable battery and a switch for microphone
Add a DIP switch to disable microphone that will close #18:

[image]

Add an optional connector for battery that will close #19:

[image]
closed 2024-11-03 by mariobehling