fossasia/badgemagic-hardware #20: CH583, CH582M or CH582 build in? lsusb reports CH583

froscon2024, 2024-09-22:
What chipset is build into the hardware revision @mariobehling brought to froscon 2024? In lsusb/dmesg it reports with the preinstalled firmware to be a CH583.
The parts list is telling to be a CH582M https://github.com/fossasia/badgemagic-hardware?tab=readme-ov-file#parts-list
The firmware page is telling to be a CH582 https://github.com/fossasia/badgemagic-firmware?tab=readme-ov-file#chipset

kienvo, 2024-09-23:
CH583 is CH583M for short =], the same for CH582. Comparison table from the datasheet:
[image]

CH582 and CH583 are almost identical. The difference is the CH583 has an additional SPI and 1.7v min. supply voltage. In most cases, they can be interchangeable.

froscon2024, 2024-09-23:
Thanks for the information. 
What does your preinstalled OEM closed source firmware say about the build in chipset in lsusb?

kienvo, 2024-09-23:
Both my USB-C version and Micro-USB version, lsusb said 'CH583', but the chip is CH582.

froscon2024, 2024-09-25:
I flashed some hours ago on all badges i have the foss firmware. I checked on the chip and it was always WCH labeled CH582M.

closed 2024-09-25 by froscon2024