fossasia/badgemagic-hardware #21: Do not use the suppier of the yellow badges any more - bad build quality, a not working LED and missing part on PCB

froscon2024, 2024-09-25:
I have badges with different colors from Froscon2024. The yellow one looks really bad. One of the LED's simply just do not work. I opened it to flash the new FOSS firmware and compared to other ones its really bad overall. Even the used solderpaste they used to solder on the battery is the really worse one. The one that leave behind not a shiny surface but a milky mess. Even after i heated it up twice (disconnect and connect of battery to flash firmware) the solder stayed like that. I know this type of solder and its always used on really, really crappy products.
The PCB also miss a part that is not missing on the other ones. Its the SMD part directly next to "R3" that is on the bottom of the PCB next to the battery.
And the worst part is one of the yellow LED's is not working in the middle of the badge. It does not matter if closed source firmware or the new released opensource one. It always stay dark, break the letters i try to write and just look bad.
The PCB number of the yellow badge is B1144_220515. Its also missing the production date the other have on the top right next to the PCB number.

mariobehling, 2025-01-09:
Hi @froscon2024 we are still looking at options for future production batches. Even though it is already some time ago, could you take maybe a picture of the issue with the yellow LED?

froscon2024, 2025-08-25:
I think this is solved now. The badge with this production issues have been shown and discussed in person to @mariobehling

closed 2025-08-25 by froscon2024